youseetoo/uc2-esp32 #29: {"task":"/state_get"} gave no information in logs

jsmithe2, 2024-05-23:
I downloaded the firmware for a wemos D1 R32 board using you web interface. I followed the state_get function on the logs and it didnt provided any information like what is expected from your youtube video. What could potentially be my problem here?

jsmithe2, 2024-05-28:
@beniroquai I bought a new wemos D1 R32 board thinking that maybe the firmware wasn't downloading to the board but my second board has the same problem. I download the firmware but I get no information back when using that command. The light on the boards works and the steppers twitch when I start downloading the firmware which makes me believe the steppers connections are fine. Any troubleshooting I can do here?

beniroquai, 2024-05-29:
What'S the baudrate you'Re using?

jsmithe2, 2024-05-29:
@beniroquai I changed the baudrate to 115200 like it said but still no communication. I also tried the default 500000. I previously programmed the first board with the UC2-Rest firmware through Arduino IDE and it seemed to work but there was some code missing in the repository to move forward and you said you no longer supported it. Any other possibility why I don't have communication? Also, is the ESP32 Dev UC2 standalone board a custom-made board?

beniroquai, 2024-05-29:
This is indeed hard to debug over the air. Which program do you use for getting the Serial prints? If you were able to flash the firmware successfully using this tool: https://youseetoo.github.io/ then you should also be able to communicate. Can you try the Arduino IDE eventually? The online terminal may not work / give outputs. Could you try compiling the firmware using platform-io?  I think someone was able to do that here: https://github.com/openUC2/UC2-REST/issues/77

jsmithe2, 2024-06-18:
Hey @beniroquai, I was able to upload the firmware on a WEMOS D1 R32 board through platform.io successfully based on the instructions. I am new to serial monitors like this I am now not sure how to send the device commands and how to connect it to imswitch. From, what I understand, I can take json commands from json_api_BD.txt and add it to the main.cpp file? The directions are a little vague after the uploading steps. Where should I go from here? Is there more specific instruction elsewhere?

beniroquai, 2024-06-23:
You could try the following webpage to test the board: https://youseetoo.github.io/indexWebSerialTest.html
Connect to your ESP and then try it out :) 

What would be a good way to structure the Documentation? Did you have a look here https://openuc2.github.io/docs/Electronics/uc2e1
It's a little outdated, but @christiankuttke is trying to update some portions here and there. Would be great to see what'S necessary to involve others, too.

closed 2024-08-20 by beniroquai