youseetoo/uc2-esp32 #60: Generalize to more motors

beniroquai, 2025-02-17:
@KillerInk what would be the best way to generalize the entire code (mostly for the CAN world) to have motor motors than just 4 - say 10. Would this even be possible?

KillerInk, 2025-02-18:
Splitt Master and slave code.
Let Master scan on startup for slaves and get from them their Id or state, what it is representing.
That way Master don't have to care what slave is and can handel up to max device count Can can have

beniroquai, 2025-02-18:
I would already be happy to have more than 4 axes. So, say 10 maximum .. what would be needed to be changed? All array entries where I see a four I suppose? ;)

beniroquai, 2025-02-18:
I have uploaded something to the betabd branch - does this make sense?