youseetoo/uc2-esp32 #72: inconsistent API responses between different controls

jacopoabramo, 2025-05-22:
I'm currently controlling via serial a V3 board using the latest firmware. The below information was obtained from the [web-serial test tool](https://youseetoo.github.io/indexWebSerialTest.html).

Firmware information obtained via `/state_get` is the following:

```
{
    "state": {
        "identifier_name": "UC2_Feather",
        "identifier_id": "V2.0",
        "identifier_date":"Apr 16202512: 34: 30",
        "identifier_author":"BD",
        "IDENTIFIER_NAME":"uc2-esp",
        "configIsSet":0,
        "pindef":"UC2_3",
        "I2C_SLAVE":0
    },
    "qid":0
}
```

An example of exchange between client and device for controlling the laser output below.

Command: 
```json
{"task": "/laser_act", "LASERid": 1, "LASERval": 0}
```
Response:
```
++
{"qid":0,"success":0}
--
```

Now, an exchange between client and device but for controlling stepper motors.

Command:
```json
{"task":"/motor_act","motor":{"steppers":[{"stepperid":1,"position":1000,"speed":15000,"isabs":0,"isaccel":0}]}}
```

Response:

```
++
{"qid":0,"success":0}
--
++
{"steppers":[{"stepperid":1,"position":1000,"isDone":0}],"qid":0}
--
```

From the [documentation](https://openuc2.github.io/docs/Electronics/Advanced/UC2-ESP/APIDescription/Laser#example-requests), the laser response should return `{"success" : 1}`, while in this case it returns the `qid` as well and the `success` flag is still set to 0.

Same thing for the motor control.

The `qid` I imagine is simply an error in the documentation; as for the `success` flag I'm not sure if it is the intended behavior or there's a bug in the latest firmware release.

Funny thing: yesterday (21st of May 2025) the `success` flag for the laser control was working according to the documentation (a.k.a. `success` flag set to 1), now it doesn't seem to work anymore. I also tried refreshing the firmware once more but to no avail.

KillerInk, 2025-05-26:
i would not rely on qid and success flag. treat it more like set and expect its successfull.
if we check this part from serialize
https://github.com/youseetoo/uc2-esp32/blob/07f1282e3e9d3a3db8b2be35ddd670fe92d4ac70/main/src/serial/SerialProcess.cpp#L187-L220

we see that success is based on the qid. but the qid itself is just something that increments on every call client side. at least it should
so if qid <0 it returns -1, qid  = 0 it returns 0 and qid > 0 returns 1.

for me thats all useless^^ but i got told its historical grown and is as it is.

about motor your are right it should only return the qid and success.
but i think it got added to avoid another call only to get new data.

jacopoabramo, 2025-05-26:
> Funny thing: yesterday (21st of May 2025) the success flag for the laser control was working according to the documentation (a.k.a. success flag set to 1), now it doesn't seem to work anymore. I also tried refreshing the firmware once more but to no avail.

This is my fault then: I set `qid = 0` in my command message on my client side so I guess this is the expected outcome.

beniroquai, 2025-05-26:
> for me thats all useless^^ but i got told its historical grown and is as it is.

Peace over QID xD

beniroquai, 2025-05-26:
> if anything, if this is historical, it could be a good opportunity to rework this or at least give it some thought

Amen on that :D 
Who is keen on the rework @ethanjli? ;-) 

It only gets negative if there is no Module associated in the main application (e.g. not compiled or simply wrong json string..).

beniroquai, 2025-05-26:
> about historical its because of imswitch. there the parser need that ++++ ---- to identifiy the message start and end and the qid identifys wich module it cam from. but with json and and the module string inside, is this theortical obsolete.

This is true. Well, probably also historically.. I think this could be handled differenty (?). Back then I needed a delimiter, but since every response has only one line now, it should be easier to parse the jsons even without hte ++--
https://github.com/openUC2/UC2-REST/blob/master/uc2rest/mserial.py#L353-L380

Shall we switch?

jacopoabramo, 2025-05-28:
If you want something to decode the incoming JSON messages I would reccomend [`msgspec`](https://jcristharif.com/msgspec/). I'm currently using it to write some dataclasses (in msgspec they're called `Struct`) to encode messages in binary JSON format and also decode the responses from the serial. It's pretty handy, easy to use and very fast to do both encoding and decoding.

beniroquai, 2025-05-29:
We already discussed that alongside protobuffer together with @KillerInk and @ethanjli. I would like to keep json for higher level stuff, but for speed msgpack and alternatives would be key. Would probably need a wrapper/parser on both sides, or?

jacopoabramo, 2025-05-29:
Good question. I'll open a separate issue for it so I can elaborate