youseetoo/uc2-esp32 #78: feature: `issync` flag in API commands

jacopoabramo, 2025-06-06:
I had a chat with @beniroquai today and he explained me a little bit better the communication model of the firmware.

As I understand, the idea is that the client should provide an asynchronous type of handling where there's a dedicated `RX` buffer which constantly polls the serial object for new responses incoming from the motor for continous, event-driven acquisition.

In my case I'm going for a more simpler approach where I would expect that having sent a command, my client would expect a confirmation response (including the possible success/fail of the operation). According to what @beniroquai suggested an initial approach would be to provide a `issync` flag (which should default to `False`) and it can be set to `True` when the client is requesting an immediate response.

This is useful for situations where an UI is sending a command and expecting a response in return so that it can validate the new position and update the view accordingly.

beniroquai, 2025-06-08:
Thanks! My question would be: Shall we have it as an additional parameter to each json communication (similar to QID) or shall we have a global flag that switches it off (persistently or until next boot). An additional flag is probably the way to go I guess, or?

jacopoabramo, 2025-06-09:
From my perspective, whatever its easier to deal on the  firmware side. I can adapt to the condition that doesn't complicate development

beniroquai, 2025-06-11:
After your latest update on PySerial side, do you think this is still relevant? Asking for a friend ;-)